rust-embedded/svd2rust #614: An idea of using `groupName` as cfg feature

duskmoon314, 2022-06-02:
# Motivation

Recently I am working on two MCU sharing most of the peripherals. So I searched for issues to find if there were any excellent solutions and found #393. It seems hard to parse multiple SVD files and merge them into one crate automatically. Thus I came up with the idea of using `groupName` as the cfg feature to partly solve this problem. This idea has concerns and may not be a very good idea. But #393 is still open, so I want to share and discuss this idea.

# Design

In the spec of SVD, every peripheral can have a child element named `groupName`:

| child elements | description | type | occurrence |
| -|-|-|-|
|groupName | Define a name under which the System Viewer is showing this peripheral. | xs:Name | 0..1 |

The idea is to add a new argument named `--group_name_as_feature` or something shorter and meaningful. When the argument is given, then generate a `#[cfg(feature = "<group_name>"]` before the definition of peripheral if it has this child element.

That is if we have an SVD file with the following definition of a uart:

```xml
<peripheral>
  <name>UART0</name>
  <description>Universal Asynchronous Receivers/Transmitters</description>
  <groupName>Interface</groupName>
  <baseAddress>0x96000000</baseAddress>
  ...
```

Then the generated code is:

```rust
#[doc = "Universal Asynchronous Receivers/Transmitters"]
#[cfg(feature = "interface")]
pub struct UART0 {
    _marker: PhantomData<*const ()>,
}
#[cfg(feature = "interface")]
unsafe impl Send for UART0 {}
...
```

And the maintainer of this crate should add something like the below to the `Cargo.toml`:

```toml
[features]
default = ["interface"]
mcu1 = ["interface"]
mcu2 = ["interface"]

interface = []
```

# Concern

Just like discussions in #433 and #254, adding cfg may cause usability and composability problems. Maintainers should spend more time maintaining `Cargo.toml` if they enable this feature. And the only benefit I see for now is to generate a crate that supports multiple MCU after someone merge multiple SVD file by hand.

I wonder if there are any other benefits. If the benefits exist, is it worth adopting this design?

Note: I have a try to implement this feature and it seems not very hard.

burrbull, 2022-06-02:
Yeah, you need generate some `Cargo.features.toml` which can them be merged into `Cargo.toml`

duskmoon314, 2022-06-03:
> Yeah, you need generate some `Cargo.features.toml` which can them be merged into `Cargo.toml`

I think that using `toml_edit` to add the cfg feature automatically may be easier to use. Maybe generate something like:

```toml
[features]
# other features, e.g. rt = ["riscv/rt"]
group_1 = []
group_2 = []
default = ["group_1", "group_2"]
```

Advantage: There is less work to do for the maintainers of the crate. 

Disadvantage: May overwrite some feature names if accidentally use the same name with the SVD file.

duskmoon314, 2022-06-04:
> Have you seen this old PR and discussion? #254

Oh, my bad. Didn't read the section about Cargo.toml generation carefully before.

Since an SVD file may contain dozens of `groupName`, it is reasonable to generate a `Cargo.toml` like the implementation in #254. However, in doing so, svd2rust users may find it annoying if they frequently update an SVD file and overwrite the `Cargo.toml` again and again.

Besides, #254 disable all peripherals by default, which I think is not compatible with the current behavior.

therealprof, 2022-06-21:
@duskmoon314 We discussed this in todays meeting and people are quite excited about the idea and convinced that it might work nicely as proposed. Any chance you have time to follow up with a PR to implement this idea?

duskmoon314, 2022-06-22:
Sure!

I've skimmed the discussion briefly. I have to say that I have not yet thought carefully about how to solve the problem of multiple peripherals with the same name. 😢

commit 69306b67 (2022-07-07): feat: add argument feature_group

- When off, do nothing else
- When on
  - generate cfg attribute for every `groupName`
  - generate `features.toml` containing all `groupName`

close #614
commit e2abc28c (2022-07-07): Merge #617

617: feat: add argument feature_group r=burrbull a=duskmoon314

Add new feature `feature_group`
- When off, do nothing else
- When on
  - generate cfg attribute for every `groupName`
  - generate `features.toml` containing all `groupName`

close #614
PR #617 (2022-07-07): feat: add argument feature_group
Add new feature `feature_group`
- When off, do nothing else
- When on
  - generate cfg attribute for every `groupName`
  - generate `features.toml` containing all `groupName`

close #614
closed 2022-07-07 by bors[bot]