rust-embedded/svd2rust #616: 0.24.0 generates wrong const-generic code

bjoernQ, 2022-06-15:
We have a register containing a field like this
```xml
            <field>
              <dim>8</dim>
              <dimIncrement>0x1</dimIncrement>
              <dimIndex>0-7</dimIndex>
              <name>CH%s_TX_THR_EVENT_INT_CLR</name>
              <description>Set this bit to clear the  rmt_ch%s_tx_thr_event_int_raw interrupt.</description>
              <bitOffset>24</bitOffset>
              <bitWidth>1</bitWidth>
              <access>write-only</access>
            </field>
```

When generating code with 0.23.0 this code produces the expected output:
```rust
    let mut data = 0u32;
    let data_ptr = &mut data as *mut _ as *mut u32;
    unsafe {
        &*(data_ptr as *mut generated23::generic::Reg<generated23::rmt::int_clr::INT_CLR_SPEC>)
    }
    .write(|w| unsafe { w.ch_tx_thr_event_int_clr(7).set_bit() });
    println!("With svd2rust 0.23.0: {:032b}", data);
```

The output is `With svd2rust 0.23.0: 10000000000000000000000000000000`

However, generating code from the same SVD with 0.24.0 the following code
```rust
    let mut data = 0u32;
    let data_ptr = &mut data as *mut _ as *mut u32;
    unsafe {
        &*(data_ptr as *mut generated24::generic::Reg<generated24::rmt::int_clr::INT_CLR_SPEC>)
    }
    .write(|w| unsafe { w.ch_tx_thr_event_int_clr::<7>().set_bit() });
    println!("With svd2rust 0.24.0: {:032b}", data);
```

outputs `00000000000000000000000010000000`

Apparently, it's ignoring the `bitOffset` of the field.

I created a repo to reproduce this: https://github.com/bjoernQ/svd2rust_test

Maybe I'm doing something obvious wrong but it looks like a bug to me.

burrbull, 2022-06-15:
Yes. Those functions are not equivalent. And I'm really surprised someone already used one of them.
const-generic version just takes bitOffset, not field number.
In 0.23 those function were both present in generated code. Numeric version always, const-generic under feature.
I've deleted first one during refactoring.
If you need it I could try to return it back, but it will take time.

bjoernQ, 2022-06-15:
Thanks for the quick reply. I will look into our code and we can most probably work around it.

The behavior of the const-generic version is at least surprising and we started using them when we upgraded to 0.24.0 and found the numeric version removed - we just assumed it's equivalent

No idea if this will cause similar confusion for others and if it would be better to re-add the old functionality