rust-embedded/svd2rust #618: Allow to use only as library

Shatur, 2022-06-29:
When working with FPGA I use this crate as a build dependency library to generate code (because my FPGA configuration changes).
But this also means that I have to pull some extra deps, such as `clap`.
I would like to request to separate executable and generation library.

Emilgardis, 2022-06-29:
This could be done by making bin deps default opt-in, and then you'll depend on svd2rust with `svd2rust = {version = "*", default-features = false}`

e.g

New feature `bin` that pulls in clap and everything else we're using in the svd2rust binary. This feature is enabled by default

this just has to be implemented, feel free to make a pr!

Emilgardis, 2022-06-29:
I don't see the issue, binary packages can specify [`required-features`](https://doc.rust-lang.org/cargo/reference/cargo-targets.html#the-required-features-field)

commit c8be313d (2022-07-01): Merge #619

619: Make binary dependencies optional r=burrbull,emilgardis a=Shatur

Users can now disable binary dependencies if they don't need them, closes #618.

I think we could also make JSON and YAML optional and keep them enabled by default. What do you think?
PR #619 (2022-07-01): Make binary dependencies optional
Users can now disable binary dependencies if they don't need them, closes #618.

I think we could also make JSON and YAML optional and keep them enabled by default. What do you think?
closed 2022-07-01 by bors[bot]