rust-embedded/svd2rust #620: Incorrect alias for bitfield writers

newAM, 2022-07-04:
`svd2rust` 0.24.0 will incorrectly alias some bitfield writers.

For example in the stm32wl in the 0.15 release of stm32-rs I see this:

```rs
#[doc = "Field `AESEN` writer - AESEN"]
pub use PKAEN_W as AESEN_W;

/* ... */

pub type PKAEN_W<'a> = crate::BitWriter<'a, u32, AHB3ENR_SPEC, PKAEN_A, 16>;
```

`AESEN_W` should not alias `PKAEN_W` in this case because it has a different bit position.

`AESEN_W` should probably look more like this:

```rs
pub type AESEN_W<'a> = crate::BitWriter<'a, u32, AHB3ENR_SPEC, AESEN_A, 17>;
```

This only occurs in 0.24.  I generated `stm32-rs` with version 0.23 and did not get this behavior.

newAM, 2022-07-04:
CC @burrbull @adamgreig

burrbull, 2022-07-04:
Bug in svd2rust. Ideal morning.
Try #621, please.

newAM, 2022-07-04:
That fix did not seem to do it unfortunately, still generates the alias.

newAM, 2022-07-04:
> That fix did not seem to do it unfortunately, still generates the alias.

Ah, sorry, I had both versions of svd2rust on my path, it's late in the day here 😅 

It does not alias anymore, but it also isn't generating enabled/disabled fields now:

```text
   Compiling stm32wlxx-hal v0.5.1 (/home/alex/git/stm32wl-hal/hal)
error[E0599]: no method named `disabled` found for struct `BitWriterRaw<'_, u32, AHB3ENR_SPEC, stm32wl::stm32wl5x_cm4::rcc::ahb3enr::PKAEN_A, BitM, 17_u8>` in the current scope
   --> hal/src/aes.rs:234:45
    |
234 |         rcc.ahb3enr.modify(|_, w| w.aesen().disabled());
    |                                             ^^^^^^^^ method not found in `BitWriterRaw<'_, u32, AHB3ENR_SPEC, stm32wl::stm32wl5x_cm4::rcc::ahb3enr::PKAEN_A, BitM, 17_u8>`
    |
    = note: the method was found for
            - `BitWriterRaw<'a, u32, stm32wl::stm32wl5x_cm4::adc::ier::IER_SPEC, ADRDYIE_A, BitM, 0_u8>`
            - `BitWriterRaw<'a, u32, stm32wl::stm32wl5x_cm4::adc::ier::IER_SPEC, EOSMPIE_A, BitM, 1_u8>`
            - `BitWriterRaw<'a, u32, stm32wl::stm32wl5x_cm4::adc::ier::IER_SPEC, EOCIE_A, BitM, 2_u8>`
            - `BitWriterRaw<'a, u32, stm32wl::stm32wl5x_cm4::adc::ier::IER_SPEC, EOSIE_A, BitM, 3_u8>`
            and 521 more types
```

burrbull, 2022-07-04:
check last commit

newAM, 2022-07-04:
Everything works now, thanks for the fast fix!

burrbull, 2022-07-04:
Fixed by #622

closed 2022-07-04 by burrbull