rust-embedded/svd2rust #625: svd2rust 0.24.1 generates duplicate definitions which don't compile

qwandor, 2022-07-08:
Hello,
I'm trying to update from `svd2rust` 0.24.0 to 0.24.1 for https://github.com/gd32-rust/gd32-rs/, but the new version is generating invalid code which doesn't compile. You can see some of the failures at https://github.com/gd32-rust/gd32-rs/runs/7249993930?check_suite_focus=true. For example:

```
error[E0592]: duplicate definitions with name `frozen`
     --> src/gd32f130/mod.rs:63410:13
      |
63410 |             pub fn frozen(self) -> &'a mut W {
      |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ duplicate definitions for `frozen`
...
63659 |             pub fn frozen(self) -> &'a mut W {
      |             -------------------------------- other definition for `frozen`
```

The generated code in question:

```rust
#[doc = "Field `CH1COMCTL` writer - Channel 1 output compare mode"]
pub type CH1COMCTL_W<'a, const O: u8> =
    crate::FieldWriterSafe<'a, u16, CHCTL0_OUTPUT_SPEC, u8, CH1COMCTL_A, 3, O>;
impl<'a, const O: u8> CH1COMCTL_W<'a, O> {
    #[doc = "The comparison between the output compare register CHyCV and the counter CNT has no effect on the outputs"]
    #[inline(always)]
    pub fn frozen(self) -> &'a mut W {
        self.variant(CH1COMCTL_A::FROZEN)
    }
```

```rust
#[doc = "Field `CH0COMCTL` writer - Channel 0 compare output control"]
pub type CH0COMCTL_W<'a, const O: u8> =
    crate::FieldWriterSafe<'a, u16, CHCTL0_OUTPUT_SPEC, u8, CH0COMCTL_A, 3, O>;
impl<'a, const O: u8> CH0COMCTL_W<'a, O> {
    #[doc = "The comparison between the output compare register CHyCV and the counter CNT has no effect on the outputs"]
    #[inline(always)]
    pub fn frozen(self) -> &'a mut W {
        self.variant(CH0COMCTL_A::FROZEN)
    }
```

A similar issue happens in many other places as well, and for other device crates.

qwandor, 2022-07-08:
Oh, and the corresponding section of the SVD file for the example above, I think:

```xml
<field>
  <name>CH1COMCTL</name>
  <description>Channel 1 output compare mode</description>
  <bitOffset>12</bitOffset>
  <bitWidth>3</bitWidth>
  <enumeratedValues derivedFrom="CHCTL0_Output.CH0COMCTL.CH0COMCTL"/>
</field>
```

```xml
<field>
  <name>CH0COMCTL</name>
  <description>Channel 0 compare output control</description>
  <bitOffset>4</bitOffset>
  <bitWidth>3</bitWidth>
  <enumeratedValues><name>CH0COMCTL</name><usage>read-write</usage><enumeratedValue><name>Frozen</name><description>The comparison between the output compare register CHyCV and the counter CNT has no effect on the outputs</description><value>0</value></enumeratedValue><enumeratedValue><name>ActiveOnMatch</name><description>Set channel to active level on match. OxCPRE signal is forced high when the counter matches the capture/compare register CHyCV</description><value>1</value></enumeratedValue><enumeratedValue><name>InactiveOnMatch</name><description>Set channel to inactive level on match. OxCPRE signal is forced low when the counter matches the capture/compare register CHyCV</description><value>2</value></enumeratedValue><enumeratedValue><name>Toggle</name><description>OxCPRE toggles when CNT=CHyCV</description><value>3</value></enumeratedValue><enumeratedValue><name>ForceInactive</name><description>OxCPRE is forced low</description><value>4</value></enumeratedValue><enumeratedValue><name>ForceActive</name><description>OxCPRE is forced high</description><value>5</value></enumeratedValue><enumeratedValue><name>PwmMode0</name><description>In upcounting, channel is active as long as CNT&lt;CHyCV else inactive. In downcounting, channel is inactive as long as CNT&gt;CHyCV else active</description><value>6</value></enumeratedValue><enumeratedValue><name>PwmMode1</name><description>Inversely to PwmMode0</description><value>7</value></enumeratedValue></enumeratedValues>
</field>
```

burrbull, 2022-07-09:
Where can I see full SVD file?

qwandor, 2022-07-11:
Here's one: https://q.geek.nz/files/gd32f130.svd.patched

burrbull, 2022-07-11:
> Here's one: https://q.geek.nz/files/gd32f130.svd.patched

[image]

qwandor, 2022-07-11:
Whoops sorry, try now.

Emilgardis, 2022-07-23:
Fixed with #631

closed 2022-07-23 by Emilgardis