rust-embedded/svd2rust #699: Compiler compains that Peripherals is not an iterator

erazor-de, 2022-11-29:
I'm using
rustc 1.65.0 (897e37553 2022-11-02)
svd2rust 0.27.2 ( )

I made a crate from the manufacturers STM32F103.svd using your documentation. When I try to use the crate in another project like:
```
#![no_std]
#![no_main]

use panic_halt as _;

use cortex_m_rt::entry;
use stm32f103;

#[entry]
fn main() -> ! {
    let peripherals = stm32f103::Peripherals::take().unwrap();
    let cortex = cortex_m::Peripherals::take().unwrap();

    loop {}
}
```
I get the error message:
```
error: src/main.rs:11: `stm32f103::Peripherals` is not an iterator
error: src/main.rs:11: `stm32f103::Peripherals` is not an iterator
note: src/main.rs:11: the following trait bounds were not satisfied:
`stm32f103::Peripherals: Iterator`
which is required by `&mut stm32f103::Peripherals: Iterator`
error: stm32f103/src/lib.rs:2126: doesn't satisfy `stm32f103::Peripherals: Iterator`
error: could not compile `stm32f103rb` due to 2 previous errors
```
Taking the Peripherals of the cortex_m crate produces no error. I tried with edition 2021 and 2018 in Cargo.toml.

Emilgardis, 2022-11-29:
please post the entire error output instead of a truncated one.

what code is placed at `stm32f103/src/lib.rs:2126`?

Where did you get the svd file?

erazor-de, 2022-11-29:
The using crate now has two new feature uses regarding critical-section:
```
cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
stm32f103 = { version = "0.0.1", path = "./stm32f103", features = ["rt", "critical-section"] }
```
This seems to work now.

closed 2022-11-29 by erazor-de