rust-embedded/svd2rust #705: Field writer has no impl when enumeratedValue is derived

kaizensparc, 2023-01-06:
Hello, 
I'm currently implementing a PAC/HAL for the GD32E103 device and set the SVD to derive TIMER*.CTL0.CEN registers to TIMER0.CTL0.CEN register:
```
            <field>
              <name>CEN</name>
              <description>Counter enable</description>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
              <enumeratedValues derivedFrom="TIMER0.CTL0.CEN.CEN"/>
            </field>
```

The generated code reuses the CEN_A and CEN_R types from TIMER0 to every timer but not the CEN_W. The issue is that is defines the type (`pub type CEN_W<'a, const O: u8> = crate::BitWriter<'a, u32, CTL0_SPEC, CEN_A, O>;`) but it does not provides its implementation (to set the variants) as it does on TIMER0.

This issue appeared between svd2rust 0.24.0 and 0.25.0.

burrbull, 2023-01-07:
Please, add link on source

kaizensparc, 2023-01-07:
An example of the bug is available on https://github.com/gd32-rust/gd32-rs for the gd32f1 timer peripheral (the gd32e1 is still a WIP). The generated source code is available on https://github.com/gd32-rust/gd32-rs-nightlies/tree/main/gd32f1. I attached the patched svd used by svd2rust for generation. On the svd file you can see lines 17899-17905 the derived I quoted earlier.
On the generated code you can compare the two CEN_W types for TIMER0 and TIMER1 (which is derived from TIMER0). The documentation for TIMER0.CTL0.CEN is on https://docs.rs/gd32f1/latest/gd32f1/gd32f130/timer0/ctl0/type.CEN_W.html and has the implementation. The one for TIMER1.CTL0.CEN is on https://docs.rs/gd32f1/latest/gd32f1/gd32f130/timer1/ctl0/type.CEN_W.html and lacks the implementation.

[gd32f130.svd.txt](https://github.com/rust-embedded/svd2rust/files/10366421/gd32f130.svd.txt)

burrbull, 2023-06-05:
[image]

PR #731 (2023-06-13): generic REG in field writers
should fix #705
closed 2023-06-13 by therealprof