rust-embedded/svd2rust #706: Simple bitfield definition generates type error

ReinoutHeeck, 2023-01-15:
I was handcrafting some svd (yaml) and early on ran into a generator issue.

To reproduce run the following in an empty directory: 
```sh
cargo init simple_bit_field_failure \
&& cd simple_bit_field_failure \
&& cargo add cortex_m vcell \
&& cd src \
&& echo """
name: MyDevice
version: 0
description: My Device
addressUnitBits: 32
width: 32
peripherals:
  - name: MyPeripheral
    baseAddress: 0xb8000300
    registers:
      - register:
          name: MyRegister
          description: My register.
          addressOffset: 0
          size: 1
          fields:
            - name: MyBitField
              description: My bit field.
              bitRange: \"[0:0]\"
""" |  RUST_LOG=debug svd2rust --strict --source_type yaml \
&&cargo fmt \
&&cargo build
```

result:
```
...
error[E0308]: mismatched types
   --> src/lib.rs:771:37
    |
771 |                 MY_BIT_FIELD_R::new(self.bits)
    |                 ------------------- ^^^^^^^^^ expected `bool`, found `u8`
    |                 |
    |                 arguments to this function are incorrect
    |
...

```

ReinoutHeeck, 2023-01-15:
After delving into the generator sources it turns out that register 'size' above is measured in bits instead of bytes as I was using it.

This uncovers a broken corner case in the logic for computing `use_cast` in `register.rs`.

Fixing support for single-but registers is not very urgent I guess ;-)

burrbull, 2023-01-16:
no need to dive in code. there is specification https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_special.html#registerPropertiesGroup_gr

ReinoutHeeck, 2023-01-16:
I knew that site but seeing uncompilable  code generated I also 'knew' it was a bug and had to dive in. It's my knee-jerk reaction :-/