rust-embedded/svd2rust #707: Doesn't support `read/clear` register access type.

deng-yongsheng, 2023-01-21:
Hi, I am using a `CH32V307` risc-v chip, trying to generate a HAL from official svd file: `CH32V307xx.svd`. 
    Svd2rust prompts an parsing error:
```text
[INFO  svd2rust] Parsing device from SVD file
[ERROR svd2rust] Error parsing SVD XML file

    Caused by:
        0: In device `CH32V30xxx`
        1: In peripheral `ETHERNET_DMA`
        2: In register `DMASR`
        3: In field `TS`
        4: Parsing unknown access at 15756:15
        5: unknown access variant 'read/clear' found
```
    Partial svd file:
```xml
   <register>
          <name>DMASR</name>
          <displayName>DMASR</displayName>
          <description>Ethernet DMA status register</description>
          <addressOffset>0x14</addressOffset>
          <size>0x20</size>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>TS</name>
              <description>Transmit status</description>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read/clear</access>
            </field>
            <field>
              <name>TPSS</name>
              <description>Transmit process stopped
              status</description>
              <bitOffset>1</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read/clear</access>
            </field>
```
    Full content of  `CH32V307xx.svd`:
[CH32V307.svd](https://gist.github.com/deng-yongsheng/e47ed59ddfc69dc4cc29829999de6730)

burrbull, 2023-01-21:
`read/clear` is incorrect access type.
use `access: read-write`  with `modifiedWriteValues: clear`

deng-yongsheng, 2023-01-22:
> `read/clear` is incorrect access type.
> use `access: read-write`  with `modifiedWriteValues: clear`
> or
> `access: read`  with `readAction: clear`
> depending on what `read/clear` means
> 

Okay, thank you.

closed 2023-01-23 by deng-yongsheng