rust-embedded/svd2rust #743: Unsupported clippy::missing_safety_doc lints (Rust Nightly) and removal of private_in_public

SteelPh0enix, 2023-09-13:
Recent update of Rust Nightly toolchain (13.09.2023) has seemingly enabled `clippy::missing_safety_doc` lint by default.
Since svd2rust generates it's documentation in form of macros, and does not add `# Safety` sections, it's failing `cargo check` on that lint.

Similar issue happens due to the removal of `private_in_public`, which is replaced by other checks.

While the second issue is pretty easy to resolve (i've manually replaced the deny generated by https://github.com/rust-embedded/svd2rust/blob/master/src/generate/device.rs#L63 with `deny(private_interfaces)` and `deny(private_bounds)`), and the `missing_safety_doc` can be either resolved in similar way, by silencing the check with another `allow(clippy::missing_safety_doc)`.

Would be nice to see that in svd2rust release, so we can avoid manually fixing the generated crate every time :)

Emilgardis, 2023-09-13:
the removal of the `private_in_public` lint shouldn't error, only warn, it should hit `renamed_and_removed_lints`

SteelPh0enix, 2023-09-13:
yes, it does, my bad - we're treating warnings as errors in our project, so i've mistakenly treated it as such.
Still, it's a warning that should not really be there i suppose?

PR #747 (2023-10-02): release 0.30.1
Fixes #743
closed 2023-10-02 by burrbull