rust-embedded/svd2rust #760: Unstable lints in v0.30.1

daniestevez, 2023-11-10:
svd2rust v0.30.1 is including
```
# ! [deny (private_bounds)]
# ! [deny (private_interfaces)]
```
in the generated output when running as `svd2rust --target none -i my_registers.svd -o src/`

These lints are unstable and the resulting code gives warnings when built with rustc 1.73.0:
```
warning: unknown lint: `private_bounds`
  --> src/lib.rs:11:1
   |
11 | # ! [deny (private_bounds)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: the `private_bounds` lint is unstable
   = note: see issue #48054 <https://github.com/rust-lang/rust/issues/48054> for more information
   = note: `#[warn(unknown_lints)]` on by default

warning: unknown lint: `private_interfaces`
  --> src/lib.rs:12:1
   |
12 | # ! [deny (private_interfaces)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: the `private_interfaces` lint is unstable
   = note: see issue #48054 <https://github.com/rust-lang/rust/issues/48054> for more information
```

svd2rust v0.30.0 doesn't do this. I've seen that this change was in included in https://github.com/rust-embedded/svd2rust/commit/f097443c93bc81a2f768e29b575eacbadeb0b832, apparently to fix the clippy lints on nightly, but it seems that it breaks them on stable.

Emilgardis, 2023-11-12:
I think we should add `#[allow(unknown_lints)]` to circumvent this, but also possibly `#[allow(renamed_and_removed_lints)]` (I'm less sure about this one, since hiding these would mean it's harder to catch.)

jessebraham, 2023-11-12:
> I think we should add `#[allow(unknown_lints)]` to circumvent this, but also possibly `#[allow(renamed_and_removed_lints)]` (I'm less sure about this one, since hiding these would mean it's harder to catch.)

I was going to suggest the former option, as well.

ianrrees, 2023-11-16:
I wonder if it would make sense to have the CI checks set `RUSTFLAGS="-D warnings"` so that compile warnings cause the check to fail?

ianrrees, 2023-11-16:
Thanks @burrbull !  Two takeaways:

1. The warnings in question go away with Rust 1.74.0 which recently stabilized.
2. Per discussion around RFC 3389, which added the lints table to Cargo.toml (in particular [this comment](https://github.com/rust-lang/cargo/issues/12115#issuecomment-1789162435)), it doesn't seem like the lints table currently provides a way to make cargo error out if warnings are encountered.  https://github.com/rust-lang/cargo/pull/12875 would add a cargo command line option to do that though.

PR #762 (2023-11-17): remove unstable lints
Fix #760 

I don't see real necessity in these lints. So this PR just removes them.
closed 2023-11-17 by Emilgardis