rust-embedded/svd2rust #761: Empty device.x output with MSP430?

natevw, 2023-11-13:
I used [tixml2svd](https://github.com/dhoove/tixml2svd) to generate an "msp430f4438a.svd" file for a variant that doesn't have any public PAC. With this file I then ran:

```
cargo install svd2rust
svd2rust -g --target=msp430 -i msp430f5438a.svd
```

followed by all the post-processing/cleanup steps and gave it a Cargo\.toml based on the example:

```
[package]
edition = "2018"
name = "msp430f5438a"
version = "0.1.0"

[dependencies]
critical-section = { version = "1.0", optional = true }
msp430 = "0.4.0"
msp430-rt = { version = "0.4.0", optional = true }
vcell = "0.1.0"

[features]
rt = ["msp430-rt/device"]
```

with rust-toolchain\.toml:

```
[toolchain]
channel = nightly
components = [ "rust-src" ]
```

in a shared parent folder:

```
repo
| - msp430f54380a   # src/ inside from svd2rust
| - firmware
```

but when I depend on `path = "../msp430f5438a"` in my "firmware" (app) crate I get an error:

> ERROR(msp430-rt): .vector_table is shorter than expected.

Now when I compare the device\.x file that svd2rust gave me it is completely blank, whereas I see in other PAC that people have created using this e.g. <https://github.com/cr1901/msp430f5529/blob/18aab52/device.x>  there's a bunch of `PROVIDED` lines in theirs. Is this empty device\.x the cause of my "shorter than expected" vector table? How do I get `svd2rust` to fill in the device\.x file it creates?

I should also note that when I run it I get a bunch of "WARN" logs from `svd2rust::generate::periperhal` about

> Found type name conflict with region Some("uca"), renamed to Some("uca0ctl0")

etc. etc. for many various uca/ucb lines, if that's at all related?

My goal is simply a working PAC crate for the MSP430F5438A which I can use with the <https://github.com/rust-embedded/msp430-quickstart> template, unfortunately I'm learning a lot of things all at once here and perhaps out of my depth hoping svd2rust would be able to automatically create one for me?

natevw, 2023-11-13:
Closing this, sorry for the noise.

When I generate an initial SVD with [msp430_svd](https://github.com/pftbest/msp430_svd) instead, the device\.x file does get some content.

(The end-result build of my app crate still has same error, but I'm guessing now that's maybe more directly just because it has hardcoded values for a different MSP430 — on closer inspection there seems no direct relation between the device\.x in the PAC and the memory\.x in the quickstart app beyond the file extension coincidence.)

closed 2023-11-13 by natevw

Emilgardis, 2023-11-13:
the device.x contains the interrupts, memory.x contains the memory layout

memory.x is included [here](https://github.com/rust-embedded/msp430-rt/blob/0eedf021d5c59ebf5434745f7882e0b09c52b070/link.x.in#L1), see [this](https://github.com/rust-embedded/msp430-rt/blob/0eedf021d5c59ebf5434745f7882e0b09c52b070/src/lib.rs#L37) for more information

device.x is included if `msp430-rt` feature `device` is enabled [here](https://github.com/rust-embedded/msp430-rt/blob/0eedf021d5c59ebf5434745f7882e0b09c52b070/build.rs#L28)

natevw, 2023-11-13:
Thanks! I think I got it working with memory\.x in my firmware app crate:

```
MEMORY
{
  RAM : ORIGIN = 0x1C00, LENGTH = 0x4000

  /*
  n.b. there's another much bigger section of ROM (i.e. Flash) on this MSP430!
  FLASH2                  : origin = 0x10000,length = 0x35C00
  */
  ROM : ORIGIN = 0x5C00, LENGTH = 0xA380

  VECTORS : ORIGIN = 0xFF80, LENGTH = 0x80
}

/* Stack begins at the end of RAM:
   _stack_start = ORIGIN(RAM) + LENGTH(RAM); */
```

Not sure how to include the rest of the Flash (what Rust seems to call "ROM" afaict) section but otoh not sure if we'll even need that for this project yet 🤷