rust-embedded/svd2rust #780: Output Cargo.toml dependencies and features

Emilgardis, 2023-11-29:
Should we have a way to make `svd2rust` output a `[dependencies]` and `[features]` given a set of arguments?

```
svd2rust cargo-toml --impl-defmt defmt -i STM32F30x.svd
```

would give

```toml
[dependencies]
critical-section = { version = "1.0", optional = true }
cortex-m = "0.7"
cortex-m-rt = { version = "0.7", optional = true }
vcell = "0.1"
defmt = { version = "0.3", optional = true }

[features]
rt = ["cortex-m-rt/device"]
defmt = ["dep:defmt"]
```

Emilgardis, 2023-11-29:
This is partly motivated by #728 where I'd want this

burrbull, 2023-11-29:
how you suggest to chose versions?

"*" ?

burrbull, 2023-11-29:
maybe then some `--init-crate name` instead that creates full-featured crate which we can just `cargo build`?

Emilgardis, 2023-11-29:
Yeah, I think using `*` in version could work, but there have been times (long time ago) where svd2rust didn't target latest release of a dependency. Doesn't happen often though, so `*` would be fine for me

Emilgardis, 2023-11-29:
I don't think a full `Cargo.toml` is good enough, since the changes you would do to the toml file wouldn't persist between runs.