rust-embedded/svd2rust #798: Versions v0.31.2-4 generates invalid code for cluster in bl602-pac

9names, 2024-01-04:
Using [this svd](https://github.com/sipeed/bl602-pac/blob/main/soc602_reg.svd) and svd2rust version 0.31.2, 0.31.3, or 0.31.4 generates a PAC that fails to build.
0.31.1 and 0.30.3 do generate a PAC that can successfully build.
This is the error returned:

```text
error[E0373]: closure may outlive the current function, but it borrows `self`, which is owned by the current function
   --> src/dma.rs:109:20
    |
109 |         (0..4).map(|n| unsafe {
    |                    ^^^ may outlive borrowed value `self`
110 |             &*(self as *const Self)
    |                ---- `self` is borrowed here
    |
note: closure is returned here
   --> src/dma.rs:109:9
    |
109 | /         (0..4).map(|n| unsafe {
110 | |             &*(self as *const Self)
111 | |                 .cast::<u8>()
112 | |                 .add(256)
113 | |                 .add(256 * n)
114 | |                 .cast()
115 | |         })
    | |__________^
help: to force the closure to take ownership of `self` (and any other referenced variables), use the `move` keyword
    |
109 |         (0..4).map(move |n| unsafe {
    |                    ++++
```

here is the generated code that it is referring to (or you can view it [here](https://github.com/9names/bl602-pac/blob/dfac4f8413d79b0707a02af4229374901db110ce/src/dma.rs#L108C26-L108C26)):
```rust
#[doc = "Iterator for array of:"]
    #[doc = "0x100..0x150 - Cluster CH%s, containing SrcAddr, DstAddr, LLI, Control, Config"]
    #[inline(always)]
    pub fn ch_iter(&self) -> impl Iterator<Item = &CH> {
        (0..4).map(|n| unsafe {
            &*(self as *const Self)
                .cast::<u8>()
                .add(256)
                .add(256 * n)
                .cast()
        })
    }
```

applying the fix in that error message is sufficient to get the PAC to build.

the cluster that is causing the issues is on [line 24279 of the svd](https://github.com/sipeed/bl602-pac/blob/64835f6349fc52e526b197be9e983c8718c29356/soc602_reg.svd#L24279), in case this is specific to the way the cluster is written.

burrbull, 2024-01-04:
Make PR, please.

See https://github.com/rust-embedded/svd2rust/blob/6b14520c9f2da3eed798ef8ef09304386f17b6a8/src/generate/peripheral/accessor.rs#L193

closed 2024-01-04 by burrbull

burrbull, 2024-01-04:
@9names you could try also to run `svd2rust` with `--max-cluster-size` option. This should make code cleaner if clusters don't have intersections with other registers.

PR #799 (2024-01-04): fix missing move in register/cluster array iter closure
fixes #798

same fix as done in #789 which was caused by #775