rust-embedded/svd2rust #801: What's the next?

nan-mu, 2024-01-17:
I am a college student, learning how to write a Rust HAL library for a Cortex-M0 microcontroller. I have built the PAC library using svd2rust. Then I wanted to write a blink program based on the programming examples in the manual of the microcontroller (I can ensure that I follow the requirements completely). 

The compilation process did not have any errors, but the microcontroller failed to execute the commands as expected. 

Did I miss a step? I saw that regardless of whether or not you use the HAL library, you need `use hal as _;`, what's the magic behind it?

nan-mu, 2024-01-17:
I will apologize if this isn't the ideal place to ask this question. 
Or could you please advise me where I might find a more suitable forum?

nan-mu, 2024-01-17:
Just in case, my code:
```rust
#![no_std]
#![no_main]
use cortex_m_rt::entry;
use cw32f030_hal as hal;
use panic_halt as _;

#[entry]
fn main() -> ! {
    let dp = hal::svd::Peripherals::take().unwrap();
    //dp.SYSCTRL.hsi.write(|w| w.div().variant(0b0101));
    dp.SYSCTRL.ahben.write(|w| w.gpioc().bit(true)); //Turn on the clock on the AHB bridge, GPIOC on the AHB, 1 for on
    dp.GPIOC.lock.write(|w| w.pin13().bit(false)); //Open the GPIOC register to operate the lock, 1 is the lock and 0 is the open
    dp.GPIOC.analog.write(|w| w.pin13().bit(false)); //Set PC13 to digital, 0 to digital, 1 to analog
    dp.GPIOC.dir.write(|w| w.pin13().bit(false)); //Set PC13 as output, 0 as output, and 1 as input
    dp.GPIOC.opendrain.write(|w| w.pin13().bit(true)); //Set the PC13 output mode, 0 is the push-pull output, and 1 is the open-drain output
    dp.GPIOC.driver.write(|w| w.pin13().bit(true)); //Set the output drive capability of PC13. 0 is a high drive and 1 is a low drive
    dp.GPIOC.speed.write(|w| w.pin13().bit(false)); //Set the output speed of PC13, 0 is low speed, 1 is high speed
    dp.GPIOC.odr().write(|w| w.pin13().bit(false)); //Set the PC13 output level, 0 is low, 1 is high
    loop {}
}
```

Emilgardis, 2024-01-17:
Hi!

> I saw that regardless of whether or not you use the HAL library, you need `use hal as _;`, what's the magic behind it?

There is no magic here, and I'm not sure what you mean as your code doesn't do that. However, looking at https://docs.rs/cw32f030-hal/0.1.1/cw32f030_hal/ I see some things not aligning with how we usually make PAC and HAL crates. The [PAC crate](https://docs.rust-embedded.org/book/start/registers.html#using-a-peripheral-access-crate-pac) is usually its own crate, not a module in the [HAL crate](https://docs.rust-embedded.org/book/start/registers.html#using-a-hal-crate).

A HAL crate is not created by svd2rust, rather it's made manually and implements the [embedded-hal](https://docs.rs/embedded-hal/latest/embedded_hal/index.html) abstractions and traits for the specific target.

You say that the code doesn't work as expected. Did you _flash the binary_ to the microcontroller?
Also, what do you expect the code to do? Right now, it seems to just do some setup.

You can reach out to us on https://matrix.to/#/#rust-embedded:matrix.org for help with this.

Some general tips for you also:

1. separate the PAC and HAL layers. The PAC crate provides direct access to peripheral registers, and the HAL layer abstracts these details through higher-level APIs.
2. You can depend on other unpublished crates via [`path = "./my-crate"`](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-path-dependencies). You can use this fact to easily iterate over your design.

nan-mu, 2024-01-17:
Thank you for pointing out the difference between the PAC and HAL libraries. I will correct the name in the future.

I have learned how to create a PAC library from the svd2rust guide, but I have not found a suitable tutorial for writing a HAL library. Does one exist?

I am confident that the compiled file and firmware created by objcopy with the ARM toolchain were burned into the microcontroller using the correct flash application (my probe program has the ability to read the entire memory).

The configuration made by the program is as shown in the comments, from the chip compilation example (following 9.4.1 of https://www.whxy.com/uploads/files/20231212/CW32x030_UserManual_EN_V1.0.pdf, as shown below), which normally outputs a low level on pc13 to light up an onboard LED. I have verified that the LED is not damaged by making the pin low in an inappropriate way.

[image]

Thank you again for your answer!

closed 2024-11-05 by burrbull