rust-embedded/svd2rust #811: thread 'main' panicked at src/util.rs:95:9: "...%s" is not a valid Ident

BrokenR3C0RD, 2024-02-14:
I'm working on patching and fixing up the SVD for the LPC11U1x/2x/3x series of MCUs. One of the modifications I made was to use `derivedFrom` in more places (so that the types used could also be used everywhere), and another was to add a generic set of IOCON registers. [Part of the SVD in question](https://gist.github.com/BrokenR3C0RD/9fa8a1697b5bca9c1ab81221c33a688f)

`<field derivedFrom="IOCON.PIO0[%s].MODE">`, however, causes svd2rust to panic with `pio0%s is not a valid Ident`. I attempted to resolve this on `master` by doing a patch similar to the following
```diff
diff --git a/src/generate/register.rs b/src/generate/register.rs
index 21fb174..d0fe56b 100644
--- a/src/generate/register.rs
+++ b/src/generate/register.rs
@@ -1462,7 +1462,7 @@ fn base_syn_path(
     } else if base.register().block == fpath.register().block {
         let mut segments = Punctuated::new();
         segments.push(path_segment(Ident::new("super", span)));
-        segments.push(path_segment(base.register().name.to_snake_case_ident(span)));
+        segments.push(path_segment(replace_suffix(&base.register().name, "").to_snake_case_ident(span)));
         segments.push(path_segment(base_ident.clone()));
         type_path(segments)
     } else {
```

This leads to `svd2rust` not panicking, though it does produce invalid code for the IOCON RegisterBlock (return type of `PIO10` instead of `PIO1` for the `pio1` method, `&PIO10` instead of `&PIO1` for `pio1_iter`). I did a similar patch to `case-defaults` (#805), and this resulted in proper codegen.

`RUST_BACKTRACE=1` panic:
```
thread 'main' panicked at src/util.rs:95:9:
"pio0%s" is not a valid Ident
stack backtrace:
   0: rust_begin_unwind
             at /rustc/f688dd684faca5b31b156fac2c6e0ae81fc9bc90/library/std/src/panicking.rs:645:5
   1: core::panicking::panic_fmt
             at /rustc/f688dd684faca5b31b156fac2c6e0ae81fc9bc90/library/core/src/panicking.rs:72:14
   2: proc_macro2::fallback::validate_ident
             at /home/masterr3c0rd/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.71/src/fallback.rs:828:9
   3: proc_macro2::fallback::Ident::new_checked
             at /home/masterr3c0rd/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.71/src/fallback.rs:760:9
   4: proc_macro2::imp::Ident::new_checked
             at /home/masterr3c0rd/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.71/src/wrapper.rs:646:50
   5: proc_macro2::Ident::new
             at /home/masterr3c0rd/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.71/src/lib.rs:956:21
   6: svd2rust::util::ToSanitizedCase::to_snake_case_ident
             at /mnt/projects/Personal/svd2rust/src/util.rs:95:9
   7: svd2rust::generate::register::base_syn_path
             at /mnt/projects/Personal/svd2rust/src/generate/register.rs:1465:36
   8: svd2rust::generate::register::fields
             at /mnt/projects/Personal/svd2rust/src/generate/register.rs:883:37
   9: svd2rust::generate::register::render_register_mod
             at /mnt/projects/Personal/svd2rust/src/generate/register.rs:283:17
  10: svd2rust::generate::register::render
             at /mnt/projects/Personal/svd2rust/src/generate/register.rs:112:25
  11: svd2rust::generate::peripheral::render_ercs
             at /mnt/projects/Personal/svd2rust/src/generate/peripheral.rs:1355:36
  12: svd2rust::generate::peripheral::render
             at /mnt/projects/Personal/svd2rust/src/generate/peripheral.rs:240:21
  13: svd2rust::generate::device::render
             at /mnt/projects/Personal/svd2rust/src/generate/device.rs:204:22
  14: svd2rust::run
             at /mnt/projects/Personal/svd2rust/src/main.rs:272:17
  15: svd2rust::main
             at /mnt/projects/Personal/svd2rust/src/main.rs:363:25
  16: core::ops::function::FnOnce::call_once
             at /rustc/f688dd684faca5b31b156fac2c6e0ae81fc9bc90/library/core/src/ops/function.rs:250:5
```

BrokenR3C0RD, 2024-02-14:
Tested the similar patch for `case-defaults` again:
```diff
diff --git a/src/generate/register.rs b/src/generate/register.rs
index 5d83140..5b75711 100644
--- a/src/generate/register.rs
+++ b/src/generate/register.rs
@@ -1486,7 +1486,7 @@ fn base_syn_path(
         let mut segments = Punctuated::new();
         segments.push(path_segment(Ident::new("super", span)));
         segments.push(path_segment(ident(
-            &base.register().name,
+            &replace_suffix(&base.register().name, ""),
             config,
             "register_mod",
             span,
```

This produces correct code, although the accessor for `IOCON.PIO1[%s]` is named `pio10`, and uses `Pio10` as the alias (which is imported correctly, unlike `master`).

burrbull, 2024-02-25:
@BrokenR3C0RD Can we close this?