rust-embedded/svd2rust #821: warning: lint private_in_public

sepbijani, 2024-03-05:
svd2rust generates code which has following lint: 
#![deny(private_in_public)]
during compilation a warning explains that the mentioned lint has been remove:

warning: lint `private_in_public` has been removed: replaced with another group of lints, see RFC <https://rust-lang.github.io/rfcs/2145-type-privacy.html> for more information
  --> src/lib.rs:11:9
   |
11 | #![deny(private_in_public)]
   |         ^^^^^^^^^^^^^^^^^
   |
   = note: `#[warn(renamed_and_removed_lints)]` on by default

Could you please update the lint or remove it, so unnecessary warning will not appear during compilation?

burrbull, 2024-03-07:
What `svd2rust` version are you using?

sepbijani, 2024-03-08:
version 0.29. 
updating to 0.32 solved the issue.

closed 2024-03-08 by sepbijani