rust-embedded/svd2rust #830: Unclear how to use `#[interrupt]` macro without deprecated `--reexport-interrupt` flag

qwandor, 2024-03-26:
According to the documentation of `svd2rust`, [the `--reexport-interrupt` flag is deprecated](https://docs.rs/svd2rust/latest/svd2rust/#the---reexport-interrupt-flag-deprecated). However according to the documentation of `cortex-m-rt`, the [`#[interrupt]` macro](https://docs.rs/cortex-m-rt/latest/cortex_m_rt/attr.interrupt.html) should only be used via the `svd2rust` reexport, and indeed it fails to compile when used directly, with a confusing error.

How is the interrupt macro supposed to be used without the `--reexport-interrupt` flag passed to `svd2rust`? This should at least be documented clearly in the `svd2rust` documentation.

burrbull, 2024-03-26:
There have been planned some changes in `cortex-m-rt` 0.8 like moving `interrupt` in independent crate. But it is not done yet.

robamu, 2024-06-25:
I also stumbled into the issue, wondering why I could not use the `interrupt` macro, and I had to rebuild with the `--reexport-interrupt` flag.

royb3, 2024-08-14:
You can use it without the flag. But you have to import your PAC's Interrupt enum as `interrupt`.

```rust
use hal::pac::Interrupt as interrupt;
use cortex_m_rt::interrupt;
```

That's basically what the `--reexport-interrupt` flag does as well. Not sure if there's any value in doing this over using the deprecated flag for now.