rust-embedded/svd2rust #835: Equivalent of command with Rust

roby2014, 2024-04-20:
hello. 

trying to use `svd2rust` in my `build.rs`, however, I'm a bit confused with `Config`.

This is an example of command I ran to simply create the files I needed
`svd2rust -g --target=riscv -i board.svd`.

Now, how can I do this with Config and pass it to `generate`?

```rs
    let conf = svd2rust::Config {
        target: svd2rust::Target::RISCV
    };
    // Generate the svd file to a string in RAM
    let pac_file = svd2rust::generate(&svd, &conf).expect("couldn't generate file with svd2rust");
 ```
 
 is there anyway of creating a default "Config" with all the struct parameters?

burrbull, 2024-04-20:
https://github.com/esp-rs/esp-pacs/blob/76e91a3e9b48b5fbf12c54b315a393ab1f3bcdb9/xtask/src/main.rs#L208

`Config::default()`. The exception is `ident-formats`, since 0.32 it is an HashMap, so needs to be filled first.

roby2014, 2024-04-20:
Thank you, that answers my question.

closed 2024-04-20 by roby2014