rust-embedded/svd2rust #844: bug: `FieldWriter` API is broken after `v0.33.2`

rmsyn, 2024-05-28:
It's unclear looking at the changes in the code, but the code of the form:

```rust
some_periph.some_register().modify(|_, w| w.some_field().variant(VARIANT))
```

Fails with an error about:

```
    = note: the following trait bounds were not satisfied:
            `u8: IsEnum`
```

So, why is generated code getting the `IsEnum` constraint for fields not specified as an `enumeratedValues` field in the SVD?

Also, if this change is intended, and I'm now "holding it wrong", shouldn't the API get a minor version bump to indicate the breaking change?

burrbull, 2024-05-28:
could I see real code where it fails? not a pseudocode

rmsyn, 2024-05-28:
From <https://codeberg.org/weathered-steel/jh71xx-hal/src/branch/main/src/uart/serial.rs#L96-L105>:

```rust
            fn write_byte(byte: u8) -> nb::Result<(), Error> {
                // SAFETY: caller must ensure exclusive access to the UART peripheral
                let uart = unsafe { &*Self::ptr() };
                if uart.lsr().read().thre().bit_is_set() {
                    uart.thr().write(|w| w.thr().variant(byte));  // <---- This is one line that raises the unimplemented trait error
                    Ok(())
                } else {
                    Err(nb::Error::WouldBlock)
                }
            }
```

Here is the SVD definition for the `thr` field:

```xml
            <field>
              <name>thr</name>
              <description>Data to be transmitted on the serial output port (sout) in UART mode or the serial infrared output (sir_out_n) in infrared mode. Data should only be written to the THR when the THR Empty (THRE) bit (LSR[5]) is set. If in non-FIFO mode or FIFOs are disabled (FCR[0] = 0) and THRE is set, writing a single character to the THR clears the THRE. Any additional writes to the THR before the THRE is set again causes the THR data to be overwritten. If in FIFO mode and FIFOs are enabled (FCR[0] = 1) and THRE is set, x number of characters of data may be written to the THR before the FIFO is full. The number x (default=16) is determined by the value of FIFO Depth that you set during configuration. Any attempt to write data when the FIFO is full results in the write data being lost.</description>
              <bitRange>[7:0]</bitRange>
              <access>write-only</access>
            </field>
```

It looks like the breaking change happened somewhere between the `v0.32.0` and `v0.33.0` release. I also tried generating the code with `v0.33.0`, and using the `set` method on the register and field (neither worked).

Generated code is completely borked by `v0.33.x` changes.

burrbull, 2024-05-28:
use unsafe "bits" method to write raw value (avaliable on any field). or "set" method if present (field has modifiedWriteConstraint)

rmsyn, 2024-05-28:
> use unsafe "bits" method to write raw value (avaliable on any field). or "set" method if present (field has writeConstraint)

Are the `Range` SVD constraints now fully supported? For instance, I'm using the `svd` crate to generate the SVD, but before adding range constraints had no effect on generated code.

I'd prefer to use a safe API, especially on fields where the full range provided by the type is safe. I'll experiment with adding a `writeConstraint` to fields.

burrbull, 2024-05-28:
> Are the `Range` SVD constraints now fully supported? For instance, I'm using the `svd` crate to generate the SVD, but before adding range constraints had no effect on generated code.

It shold looks like:
```xml
            <field>
              <name>thr</name>
              <description>Data to be transmitted on the serial output port (sout) in UART mode or the serial infrared output (sir_out_n) in infrared mode. Data should only be written to the THR when the THR Empty (THRE) bit (LSR[5]) is set. If in non-FIFO mode or FIFOs are disabled (FCR[0] = 0) and THRE is set, writing a single character to the THR clears the THRE. Any additional writes to the THR before the THRE is set again causes the THR data to be overwritten. If in FIFO mode and FIFOs are enabled (FCR[0] = 1) and THRE is set, x number of characters of data may be written to the THR before the FIFO is full. The number x (default=16) is determined by the value of FIFO Depth that you set during configuration. Any attempt to write data when the FIFO is full results in the write data being lost.</description>
              <bitRange>[7:0]</bitRange>
              <access>write-only</access>
              <writeConstraint>
                <minimum>0</minimum>
                <maximum>0xff</maximum>
              </writeConstraint>
            </field>
```

burrbull, 2024-05-28:
You could use `svdtools patch` if you do not want to edit SVD manually.

Also try new `-f` option of `form` tool.

burrbull, 2024-05-28:
Closing as this behavior is expected.

closed 2024-05-28 by burrbull

rmsyn, 2024-05-29:
> You could use svdtools patch if you do not want to edit SVD manually.

I am generating the SVD programmatically with [svd-generator](https://codeberg.org/weathered-steel/svd-generator), but thanks for the recommendation.