rust-embedded/svd2rust #849: Partial escape of HTML link from peripheral description

hegza, 2024-06-10:
Given an SVD with a HTML link in a description creates either a correct or a partially escaped link definition depending on where it is generated in the PAC.

E.g., relevant part of input SVD:

```xml
<peripheral>
	<name>Google</name>
	<version>1.0</version>
	<description>
		Link to Google: &lt;https://google.com&gt;
	</description>
	<registers>...</registers>
</peripheral>
```

Relevant parts of output:

```rust
#[doc = "Link to Google: &lt;https://google.com>"]
pub mod google;
#[doc = "Link to Google: <https://google.com>"]
pub struct Google {
    _marker: PhantomData<*const ()>,
}
```

We see the generated link symbols are partially escaped when used in the `mod` documentation and correctly escaped when used in peripheral `struct` documentation.

## Reproduction

- Attached, minimal SVD to reproduce issue: [test.zip](https://github.com/user-attachments/files/15760435/test.zip) (in Zip due to GitHub file-type restriction)
- Command to generate: `svd2rust --target riscv -i test.svd && form -i lib.rs -o src && rm lib.rs && cargo fmt`
- `svd2rust --version`
  > svd2rust 0.33.3 ( )

burrbull, 2024-06-12:
I've never see links in description. Is this allowed by SVD? I'm not sure.

hegza, 2024-06-15:
I would've assumed that svd2rust does not do transformations of the field contents but it does seem to translate special HTML characters. Surely it's wrong that it sometimes does it and sometimes not.

The behavior is relevant for our use case. I'll try to find time to investigate.

burrbull, 2024-06-16:
PRs are appreciated.

burrbull, 2024-10-20:
I've found that causer is [encode_text_minimal](https://github.com/rust-embedded/svd2rust/blob/29bcfd8695314741f12a6796e702008df22e062f/src/util.rs#L183).
Looks like removing it fixes your issue. The question is can we remove it entirely? Does it needed yet?

hegza, 2024-10-20:
Sorry, I don't necessarily understand the question. I'm also sorry I didn't get to investigating it myself. It's a low priority issue for us at the end of the day and we re-generate PACs infrequently.

We use peripheral level links to point to the relevant hardware implementation / documentation when working with open-source hardware IP.

Consistency is good enough, i.e. both links generated the same way. We can script the rest if necessary. Does that answer the question?

burrbull, 2024-10-20:
The question is why #711 was introduced. Maybe it is just not actual in newer versions of Rust.

PR #872 (2024-10-25): rm html_escape
Fixes #849
Reverts #711
closed 2024-10-25 by Emilgardis