rust-embedded/svd2rust #850: Duplicate Types Building STM32U575 from svd

JacobBeningo, 2024-06-14:
When attempting to build the converted STM32U575.svd file into a PAC, I'm getting build errors related to duplicate types. 

I'm using:
- rustc 1.79.0 (129f3b996 2024-06-10)
- svd2rust 0.33.3
- STM32U575.svd which can be found on the ST website at: https://www.st.com/content/st_com/en/search.html#q=stm32u5%20svd-t=resources-page=1

I did find that the enum size for the following fields needed to be adjusted:
TIM1:TIM1_SMCR:SMS1
TIM1:TIM1_SMCR:TS1
TIM1:TIM1_SMCR:SMS2
TIM1:TIM1_SMCR:TS2

I've notified ST about the issue. 

The main errors popping up appear to be:

1) error[E0004]: non-exhaustive patterns: `&gpdma_c13tr1::Pam::B0x0` and `&gpdma_c13tr1::Pam::B0x1` not covered
   --> src/gpdma1/gpdma_c13tr1.rs:158:23
    |
158 | #[derive(Clone, Copy, Debug, PartialEq, Eq)]
    |                       ^^^^^ patterns `&gpdma_c13tr1::Pam::B0x0` and `&gpdma_c13tr1::Pam::B0x1` not covered
    
  2) error[E0592]: duplicate definitions with name `b_0x0`
   --> src/gpdma1/gpdma_c14tr1.rs:236:5
    |
226 |     pub fn b_0x0(self) -> &'a mut crate::W<REG> {
    |     ------------------------------------------- other definition for `b_0x0`
...
236 |     pub fn b_0x0(self) -> &'a mut crate::W<REG> {
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ duplicate definitions for `b_0x0`

burrbull, 2024-06-14:
Enums generated by ST are broken and have no sense. So we just delete them before pass to `svd2rust`.
https://github.com/stm32-rs/stm32-rs/blob/e84479900343fef5f3a4f21a990dcc90a2462565/devices/stm32u575.yaml#L3

You can find generated crates here: https://github.com/stm32-rs/stm32-rs-nightlies/tree/master/stm32u5

burrbull, 2024-06-14:
Also there is some unmerged pathches: https://github.com/stm32-rs/stm32-rs/pull/974

JacobBeningo, 2024-06-14:
Awesome! Thanks for the quick reply. 

I've done this with many other chips without any issues, including the L475, so it threw me for a loop! 

I'll go this route! Thanks again!

closed 2024-11-05 by burrbull