rust-embedded/svd2rust #852: Build Errors in Code Generated from Renesas RA6M3 SVD

AddisonH, 2024-06-18:
Hello, I am seeing many errors when attempting to build a PAC from a Renesas RA6M3 SVD file. I've already fixed some errors that I found in the Renesas provided file. I've attached the modified file below.

[R7FA6M3AH.svd.txt](https://github.com/user-attachments/files/15879016/R7FA6M3AH.svd.txt)

Here is a sample of the errors that I'm seeing:

<details><summary>Error E0412</summary>
<p>

```
error[E0412]: cannot find type `P40pfs` in this scope
    --> src\pfs.rs:68:15
     |
68   |     p400pfs: [P40pfs; 10],
     |               ^^^^^^ help: a type alias with a similar name exists: `P400pfs`
...
2702 | pub type P100pfs = crate::Reg<p100pfs::P100pfsSpec>;
     | ---------------------------------------------------- similarly named type alias `P400pfs` defined here
```

</p>
</details> 

<details><summary>Error E0592</summary>
<p>

```
error[E0592]: duplicate definitions with name `offset`
   --> src\generic.rs:223:5
    |
223 |     pub const fn offset(&self) -> u8 {
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ duplicate definitions for `offset`
    |
   ::: src\glcdc\tcon_tim.rs:28:5
    |
28  |     pub fn offset(self) -> &'a mut crate::W<REG> {
    |     -------------------------------------------- other definition for `offset`
```

</p>
</details> 

<details><summary>Error E0599</summary>
<p>

```
error[E0599]: the method `variant` exists for struct `FieldWriter<'a, REG, 11, Fh, Safe>`, but its trait bounds were not satisfied
   --> src\glcdc\bg_peri.rs:29:14
    |
7   | pub struct Fh(u16);
    | ------------- doesn't satisfy `Fh: generic::IsEnum`
...
29  |         self.variant(Fh::Fh)
    |              ^^^^^^^ method cannot be called on `FieldWriter<'a, REG, 11, Fh, Safe>` due to unsatisfied trait bounds
    |
   ::: src\generic\raw.rs:44:1
    |
44  | pub struct FieldWriter<'a, REG, const WI: u8, FI = u8, Safety = Unsafe>
    | ----------------------------------------------------------------------- method `variant` not found for this struct
    |
note: trait bound `Fh: generic::IsEnum` was not satisfied
   --> src\generic.rs:312:9
    |
309 | impl<'a, REG, const WI: u8, FI, Safety> FieldWriter<'a, REG, WI, FI, Safety>
    |                                         ------------------------------------
...
312 |     FI: IsEnum,
    |         ^^^^^^ unsatisfied trait bound introduced here
note: the trait `generic::IsEnum` must be implemented
   --> src\generic.rs:54:1
    |
54  | pub trait IsEnum: FieldSpec {}
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^
``` 

</p>
</details> 

Here is the full log:
[cargo_check.txt](https://github.com/user-attachments/files/15879095/cargo_check.txt)

I've tested this SVD file with several versions of svd2rust. I found that version 0.30.x works without error, but I begin to have trouble starting with version 0.31.x and up. These error logs are from the latest tagged version, v 0.33.4.

Here is my PAC that I'm working on:
[ra6m3-pac.zip](https://github.com/user-attachments/files/15879123/ra6m3-pac.zip)

Please let me know if I can offer up any more detail or assist with debugging. Thanks.

burrbull, 2024-06-18:
It looks like 2 main issues here:

1st was introduced in #692. Looks like:
```
error[E0592]: duplicate definitions with name `p400pfs`
    --> src/pfs.rs:1050:5
     |
1045 |     pub const fn p400pfs(&self, n: usize) -> &P400PFS {
     |     ------------------------------------------------- other definition for `p400pfs`
...
1050 |     pub const fn p400pfs(&self) -> &P400PFS {
     |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ duplicate definitions for `p400pfs`
```
There should be `pub const fn p40pfs(&self, n: usize) -> &P40PFS` on 1045. Register description:
```xml
                <register derivedFrom="P100PFS">
                    <dim>10</dim>
                    <dimIncrement>0x4</dimIncrement>
                    <dimIndex>0-9</dimIndex>
                    <name>P40%sPFS</name>
                    <description>P40%s Pin Function Control Register</description>
                    <addressOffset>0x100</addressOffset>
                    <size>32</size>
                    <access>read-write</access>
                    <resetValue>0x00000000</resetValue>
                    <resetMask>0xFFFFFFFD</resetMask>
                </register>
```

2nd is related to special case not covered in #767.
EnumeratedValues with only one `isDefault` value.
```xml
                        <field>
                            <name>GAIN01</name>
                            <description>Gain value of area 1Unsigned 11-bit fixed point</description>
                            <lsb>0</lsb>
                            <msb>10</msb>
                            <access>read-write</access>
                            <enumeratedValues>
                                <enumeratedValue>
                                    <name>GAIN01</name>
                                    <description>GAIN01/1024</description>
                                    <isDefault>true</isDefault>
                                </enumeratedValue>
                            </enumeratedValues>
                        </field>
```

burrbull, 2024-06-18:
cc @n8tlarsen

9names, 2024-08-03:
as per https://github.com/ra-rs/ra/issues/6 n8tlarsen isn't active with ra-rs any more.
cc @elrafoon

ulfen, 2025-01-06:
A debug build of svd2rust panics on this input with

> thread 'main' panicked at src\generate\peripheral.rs:553:19:
> attempt to subtract with overflow

Changing the row in question (row 553 of peripheral.rs) into:

`let pad = if region.offset >= last_end { region.offset - last_end } else { 0x9999 };`

generates rows like this in the RegisterBlock of pfs.rs

```
p400pfs: [P40pfs; 10],
_reserved62: [u8; 0x9999],
```

The 0x9999 sizes comes directly after a explicitly named array with a truncated type name. This looks like a feature of register arrays, defined with dim attributes, but only when the first dimIndex is 0. This is the same feature that causes the second compile **Error E0592** above by generating multiple functions with same names.

A simple way to disable this feature is to force `sequential_indexes_from0` to false in the `expand_register` function of peripheral.rs.

`let sequential_indexes_from0 = false;`

After that the panic, the explicitly named arrays with a truncated type names, and the conflicting functions are all gone. But, there is another issue with arrays defined with dim attributes. They all have the same offset, and not only when the first dimIndex is 0.

```
pub const fn p001pfs(&self) -> &P001pfs {
    unsafe { &*core::ptr::from_ref(self).cast::<u8>().add(4).cast() }
}
pub const fn p002pfs(&self) -> &P001pfs {
    unsafe { &*core::ptr::from_ref(self).cast::<u8>().add(4).cast() }
}
pub const fn p003pfs(&self) -> &P001pfs {
    unsafe { &*core::ptr::from_ref(self).cast::<u8>().add(4).cast() }
}
```

In the last else clause of the `expand_register` function of peripheral.rs there is a for loop over an expanded info array with each item named `ri`. Yet `info.address_offset` is used instead of `ri.address_offset` twice inside the loop.

**Error E0412** is about the truncated type names. They are not imported. This can be done by removing the name change in the `render` function of register.rs right after the comment about "_Rename if this is a derived array_". But, after this change the generated code does not compile due to reimport of some of the truncated types..

There is a similar rename in the `expand_register` function of peripheral.rs with the comment "_force expansion and rename if we're deriving an array that doesnt start at 0 so we don't get name collisions_", but this renamed `ty_name` is not used. Instead the truncated `ty_str` is used to form `ty`.

Keeping the rename in the `render` function of register.rs and changing the `expand_register` function of peripheral.rs such that `ty` is formed by 

`let ty = name_to_ty(ident(&ty_name, config, "register", span));`

gives long type names that are also correctly imported.

The first **Error E0592** above, about the duplication of `offset`, is a conflict between a generated function based on a enumerated field value in the svd file and, I guess, a hard coded function with the same name.

I can finally compile the files generated by svd2rust, but will probably go for the vendor C HAL instead for now.

Fluxie, 2025-03-15:
I encountered similar problems with the RA6M5 SVD file.  

The panic @ulfen mentioned seems to be triggered by overlapping register definitions in the "PFS" peripheral. The SVD defines different ways for accessing the same registers with 32-bit wide, 16-bit wide and 8-bit wide operations.  (PmnPFS/PmnPFS_HA/PmnPFS_BY)

Example of the overlap: https://github.com/Fluxie/share/blob/main/pfs_registers.svd.xml

Not sure what is the best way forward.