rust-embedded/svd2rust #858: Whitespace in SVD files it lost

thejpster, 2024-08-18:
Raspberry Pi provide SVD files with descriptions like:

```text
        <description>DW_apb_i2c address block

            List of configuration constants for the Synopsys I2C hardware (you may see references to these in I2C register header; these are *fixed* values, set at hardware design time):

            IC_ULTRA_FAST_MODE ................ 0x0 
            IC_UFM_TBUF_CNT_DEFAULT ........... 0x8 
(etc)
        </description>
```

It comes out in the Rust code as:

```rust
#[doc = "DW_apb_i2c address block List of configuration constants for the Synopsys I2C hardware (you may see references to these in I2C register header; these are *fixed* values, set at hardware design time): IC_ULTRA_FAST_MODE ....."]
```

I suspect this is because whitespace in XML is collapsed - any run of whitespace is converted into a single space character.

I tried putting the description in a `<![CDATA[ ]>` block, but a) svdtools removed it and b) even if I force it back it, svd2rust ignores it.

How can I try and make sure the whitespace in the description is carried through into the docs? In particular, I want to avoid very very very long single line descriptions appearing in https://docs.rs/rp235x-pac/latest/rp235x_pac/index.html

burrbull, 2024-10-20:
The causer is [`respace`](https://github.com/rust-embedded/svd2rust/blob/29bcfd8695314741f12a6796e702008df22e062f/src/util.rs#L151). But I do not know what should we do with it. Sometimes squashing of strings is really helpful.

thejpster, 2024-10-20:
Can it be a configuration option?

burrbull, 2024-10-20:
> Can it be a configuration option?

Yes. It was my first thought.

burrbull, 2024-10-25:
I think `respace` was introduced to avoid this situation: https://github.com/rust-embedded/svd2rust/actions/runs/11520507042/job/32072026803?pr=879#step:7:47
I also think that this is STM32-specific issue (need to check this somehow). If it is true we I can remove `respace` entirely from `svd2rust` and move this as option to `svdtools` (to apply in `stm32-rs`).