rust-embedded/svd2rust #861: `dimIndex` in arrays is ignored in indexing accessor

bugadani, 2024-09-02:
https://github.com/rust-embedded/svd2rust/blob/f864c3ef4c545b952881b186706a105b3774500b/src/generate/peripheral/accessor.rs#L68

If I have an array with `<dimIndex>2, 1, 0</dimIndex><name>REG%s</name>`, `reg(0)` will return the element with REG2, but `reg0()` will correctly grab `REG0`. I find this behaviour rather unexpected. As a HAL author I have two options to grab the correct `REG` based on it's index:

- I can either avoid using `reg(n)` and write a big match that maps `n` to `regn`
- Or I can write a big match that maps `n` to the numeric index and pass it to `reg(n)`

burrbull, 2024-10-19:
`dimIndex`es can be 
- digits from 0
- digits from any other values
- with holes
- reversed digits
- ascii  letter
- any other

If I try (I tried to do this for field arrays in older versions  of svd2rust) to handle a part of these variants and this leads to unmaintainable code and you still have not covered other variants.
So I do not plan to do this on PAC side.

closed 2024-10-19 by burrbull

bugadani, 2024-10-19:
Well the readme doesn't specify that svd2rust intentionally only handles a subset of what's possible to describe in SVD files, so I kinda assumed generating footguns is considered an issue. 🤷‍♂️

What alternatives do we have, short of duplicating svd data in (potentially unsafe) rust code?