rust-embedded/svd2rust #866: BUG#NOW

jubao4415198, 2024-10-12:
执行指令： svd2rust -i ./GD32F4xx.svd
输出如下
`[INFO  svd2rust] Parsing device from SVD file
[INFO  svd2rust] Rendering device
thread 'main' panicked at /root/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/proc-macro2-1.0.87/src/lib.rs:847:13:
unsupported proc macro punctuation character '{'
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
`
1）我用GD32F4xx.svd试了一下，会报这个错误
2）我通过https://github.com/rust-embedded/svd2rust/issues/581中的svd文件再次测试了一下，任然会出现同样的问题。
所以可以排除svd文件的问题。
软件版本：svd2rust v0.33.4

jubao4415198, 2024-10-12:
> 执行指令： svd2rust -i ./GD32F4xx.svd 1）我用GD32F4xx.svd试了一下，会报这个错误 2）我通过[https://github.com/rust-embedded/svd2rust/issues/581中的svd文件再次测试了一下，任然会出现同样的问题。](https://github.com/rust-embedded/svd2rust/issues/581%E4%B8%AD%E7%9A%84svd%E6%96%87%E4%BB%B6%E5%86%8D%E6%AC%A1%E6%B5%8B%E8%AF%95%E4%BA%86%E4%B8%80%E4%B8%8B%EF%BC%8C%E4%BB%BB%E7%84%B6%E4%BC%9A%E5%87%BA%E7%8E%B0%E5%90%8C%E6%A0%B7%E7%9A%84%E9%97%AE%E9%A2%98%E3%80%82) 所以可以排除svd文件的问题。 软件版本：svd2rust v0.33.4 输出如下
> 
> `[INFO svd2rust] Parsing device from SVD file [INFO svd2rust] Rendering device thread 'main' panicked at /root/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/proc-macro2-1.0.87/src/lib.rs:847:13: unsupported proc macro punctuation character '{' note: run with`RUST_BACKTRACE=1`environment variable to display a backtrace`
------------------------------------------------------------------
我用二分法排查了一下
似乎只要在xml文件中存在<addressOffset>标签
<addressOffset>0x0</addressOffset>
就会报这个错误
可能 这个问题与处理<addressOffset>标签的代码有关？

jannic, 2024-10-12:
That's a known issue, see https://github.com/rust-embedded/svd2rust/issues/863.
As a workaround, you can install svd2rust using `cargo install svd2rust --locked`.

jubao4415198, 2024-10-12:
> 执行指令： svd2rust -i ./GD32F4xx.svd 1）我用GD32F4xx.svd试了一下，会报这个错误 [...]

我用二分法排查了一下 似乎只要在xml文件中存在标签 0x0 就会报这个错误 可能 这个问题与处理标签的代码有关？

> 这是一个已知问题，请参阅 #863。解决方法是，您可以使用 安装 svd2rust。`cargo install svd2rust --locked`

> 这是一个已知问题，请参阅 #863。解决方法是，您可以使用 安装 svd2rust。`cargo install svd2rust --locked`

> 非常感谢您，如果您在中国的话 我会请你喝一杯瑞幸

jubao4415198, 2024-10-12:
非常感谢您，如果您在中国的话 我会请你喝一杯瑞幸

closed 2024-10-12 by jubao4415198