rust-embedded/svd2rust #911: Issue with register names containing a % sign

robamu, 2025-02-10:
I have registers like this:

```xml
      <registers>
        <register>
          <dim>32</dim>
          <dimIncrement>0x4</dimIncrement>
          <name>PORTA[%s]</name>
          <description>PORTA Pin Configuration Register</description>
          <addressOffset>0x0</addressOffset>
          <resetValue>0x00000000</resetValue>
```

```xml
        <register derivedFrom="PORTA[%s]">
          <name>PORTB[%s]</name>
          <description>PORTB Pin Configuration Register</description>
          <addressOffset>0x80</addressOffset>
          <resetValue>0x00000800</resetValue>
        </register>
```

I get the following error with the latest version of svd2rust:

```sh
va108xx-rs/va108xx on  regenerate-pac [$] is 📦 v0.4.0 via 🦀 v1.84.1
❯ svd2rust --log debug --reexport-interrupt -i svd/va108xx.svd.patched
[DEBUG svd2rust] Current svd2rust config: [....]
[INFO  svd2rust] Parsing device from SVD file
[INFO  svd2rust] Rendering device
[DEBUG svd2rust::generate::device] Rendering interrupts
[DEBUG svd2rust::generate::device] Rendering peripheral SYSCONFIG
[DEBUG svd2rust::generate::peripheral] Checking derivation information
[DEBUG svd2rust::generate::peripheral] Pushing cluster & register information into output
[DEBUG svd2rust::generate::peripheral] Pushing 31 register or cluster blocks into output
[DEBUG svd2rust::generate::peripheral] Expanding registers or clusters into Register Block Fields
[DEBUG svd2rust::generate::device] Rendering peripheral IRQSEL
[DEBUG svd2rust::generate::peripheral] Checking derivation information
[DEBUG svd2rust::generate::peripheral] Pushing cluster & register information into output
[DEBUG svd2rust::generate::peripheral] Pushing 19 register or cluster blocks into output
[DEBUG svd2rust::generate::peripheral] Expanding registers or clusters into Register Block Fields
[DEBUG svd2rust::generate::device] Rendering peripheral IOCONFIG
[DEBUG svd2rust::generate::peripheral] Checking derivation information
[DEBUG svd2rust::generate::peripheral] Pushing cluster & register information into output
thread 'main' panicked at src/util.rs:90:5:
"Portb%s" is not a valid Ident
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

burrbull, 2025-02-10:
See https://github.com/rust-embedded/svd/pull/286

robamu, 2025-02-13:
Solved with https://github.com/rust-embedded/svd2rust/pull/908 , thanks for the quick fix :)

closed 2025-02-13 by robamu