rust-embedded/svd2rust #916: Error when access field in SVD file is "write-only"

karimpanacci, 2025-02-14:
Hi, I'm trying to get svd2rust to work with this SVD file:
[HC32L110.svd.zip](https://github.com/user-attachments/files/18805396/HC32L110.svd.zip)

but it gives me the following error:

```
[INFO  svd2rust] Parsing device from SVD file
[INFO  svd2rust] Rendering device
[ERROR svd2rust] Error rendering device
    
    Caused by:
        0: can't render peripheral 'RAM', group 'No group name'
        1: can't render register 'ICLR'
        2: Field RAM.ICLR.ERRCLR is not writtable or readable
```

The contents of the SVD file are as follows:
```
<register>
    <name>ICLR</name>
    <description>desc ICLR</description>
    <addressOffset>0xC</addressOffset>
    <size>32</size>
    <access>read-only</access>
    <resetValue>0x0</resetValue>
    <resetMask>0x1</resetMask>
    <fields>
        <field>
            <name>ERRCLR</name>
            <description>desc ERRCLR</description>
            <msb>0</msb>
            <lsb>0</lsb>
            <access>write-only</access>
        </field>
    </fields>
</register>
```

I noticed that if I enter "read-write" in the access field, for example, everything continues without errors and the PAC is generated.

What do you think is the problem?

karimpanacci, 2025-02-14:
I noticed now that the ICLR register is set as read-only and instead the ERRCLR field is set as write-only, this seems to be a contradiction, I think the manufacturer made a mistake, in your opinion which is the most correct value? both write-only or both read-only?

burrbull, 2025-02-15:
> both write-only or both read-only?

I do not know. You should check reference manual or datasheet for your chip to understand.

Most correct way in this situation is to remove `access` from register and add it to each field.

karimpanacci, 2025-02-15:
@burrbull I can't find any information about this register on the datasheet, and to tell the truth, no information about the type of access on any register...

This is the datasheet:

https://unirec.ru/upload/datasheets/DS_HC32L110SeriesDatasheet_Rev2.70.pdf

In any case I think I will close the problem since it is not a svd2rust problem

closed 2025-02-15 by karimpanacci