rust-embedded/svd2rust #917: Error parsing svd file for MSPM0

DanieleNardi, 2025-02-20:
Running svd2rust on attached file I get the following error message:
```
[INFO  svd2rust] Parsing device from SVD file
[ERROR svd2rust] Error parsing SVD XML file

    Caused by:
        0: In device `MSPM0L130X`
        1: In peripheral `FLASHCTL`
        2: In register `FLASHCTL_STATMODE`
        3: Parsing field `BANKNOTINRD` at 5158:25
        4: `EnumeratedValue error: Value 2 out of range [0 - 1]
```
Executing same command with trace level log, I get the following:
```
[DEBUG svd2rust] Current svd2rust config: Config {
        target: CortexM,
        atomics: false,
        atomics_feature: None,
        generic_mod: false,
        make_mod: false,
        skip_crate_attributes: false,
        ignore_groups: false,
        keep_list: false,
        strict: false,
        feature_group: false,
        feature_peripheral: false,
        max_cluster_size: false,
        impl_debug: false,
        impl_debug_feature: None,
        impl_defmt: None,
        output_dir: None,
        input: Some(
            "MSPM0L130X.svd",
        ),
        source_type: Xml,
        log_level: Some(
            "trace",
        ),
        interrupt_link_section: None,
        reexport_core_peripherals: false,
        reexport_interrupt: false,
        ident_formats: IdentFormats(
            {
                "enum_write_name": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "WO",
                },
                "cluster": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "field_accessor": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "field_writer": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "W",
                },
                "peripheral": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "interrupt": IdentFormat {
                    case: None,
                    prefix: "",
                    suffix: "",
                },
                "enum_value": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "cluster_accessor": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "register_mod": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "register_spec": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "Spec",
                },
                "peripheral_singleton": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "register": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "enum_value_accessor": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "cluster_mod": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "peripheral_feature": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "enum_read_name": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "register_accessor": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "enum_name": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "peripheral_mod": IdentFormat {
                    case: Some(
                        Snake,
                    ),
                    prefix: "",
                    suffix: "",
                },
                "field_reader": IdentFormat {
                    case: Some(
                        Pascal,
                    ),
                    prefix: "",
                    suffix: "R",
                },
            },
        ),
        ident_formats_theme: None,
        field_names_for_enums: true,
        base_address_shift: 0,
        settings_file: None,
        settings: Settings {
            html_url: None,
            crate_path: None,
            riscv_config: None,
        },
    }
[INFO  svd2rust] Parsing device from SVD file
[ERROR svd2rust] Error parsing SVD XML file

    Caused by:
        0: In device `MSPM0L130X`
        1: In peripheral `FLASHCTL`
        2: In register `FLASHCTL_STATMODE`
        3: Parsing field `BANKNOTINRD` at 5158:25
        4: `EnumeratedValue error: Value 2 out of range [0 - 1]
```
[MSPM0L130X.zip](https://github.com/user-attachments/files/18886634/MSPM0L130X.zip)

burrbull, 2025-02-20:
Fix `BANKNOTINRD` `bitWidth`

burrbull, 2025-02-20:
PS. There are also several syntax bugs in this file:

`access: read-writeonce` -> `read-write`

`dimArrayIndex` -> `dimIndex`