rust-embedded/svd2rust #924: unsafe attribute used without unsafe

LennyPhoenix, 2025-03-14:
svd2rust seems to generate code that fails to compile on rustc v1.85.0:

```
❯ cargo build
   Compiling stm32l476 v0.1.0 (/home/lenny/Source/stm32l476)
error: unsafe attribute used without unsafe
    --> src/lib.rs:1034:3
     |
1034 | #[no_mangle]
     |   ^^^^^^^^^ usage of unsafe attribute
     |
help: wrap the attribute in `unsafe(...)`
     |
1034 | #[unsafe(no_mangle)]
     |   +++++++         +

```

Manually adding the unsafe wrapper fixes things.

The chunk in question:

```rust
#[doc = "Digital camera interface"]
pub mod dcmi;
#[no_mangle]
static mut DEVICE_PERIPHERALS: bool = false;
#[doc = r" All the peripherals."]
#[allow(non_snake_case)]
pub struct Peripherals {
    #[doc = "DAC"]
    pub dac: Dac,
    #[doc = "DMA1"]
    pub dma1: Dma1,
    #[doc = "DMA2"]
    pub dma2: Dma2,
    #[doc = "CRC"]
    pub crc: Crc,
    // ...
}
```

SVD file used: <https://www.st.com/resource/en/svd/stm32l4_svd.zip>

Emilgardis, 2025-03-14:
You need to set the edition to "2021"

LennyPhoenix, 2025-03-14:
My apologies, that sorted it! Thanks :)

closed 2025-03-14 by LennyPhoenix

Emilgardis, 2025-03-14:
Great! We should maybe consider introducing this change in the generated code, unsure of the syntax is supported on 1.85.0 though or only on 2024 edition